soyersoyer/cameractrls #2: Add resolution and FPS

sojusnik, 2022-06-18:
Would be nice, if you could add an option to select the resolution and FPS of the webcam as well.

sojusnik, 2022-06-22:
Thank you! I will check it out when there's a more user-friendly way to install your app, f.i. when the flatpak version will be released.

soyersoyer, 2022-06-22:
What OS do you use?

sojusnik, 2022-06-23:
Ubuntu 22.04

soyersoyer, 2022-06-30:
There is a flatpak version, now you can try that! :)

soyersoyer, 2022-06-30:
The current implementation uses systemd for restoring the values, unfortunately flatpak containers don't have one. A flatpak compatible solution wil be in the v0.5.0.

sojusnik, 2022-07-01:
BTW, I've played around with the settings and didn't find a way to turn the picture in 90° steps, like from widescreen to portrait. Changing pan and tilt did nothing.

soyersoyer, 2022-07-01:
Without a real PTZ camera, pan and tilt adjustments are virtual, usually only work when zoomed. Do you have an usb camera? The UVC driver doesn't support the [V4L2_CID_ROTATE](https://elixir.bootlin.com/linux/latest/A/ident/V4L2_CID_ROTATE). I think the USB Video Class lacks rotation. Maybe you can do the same with your capture software.

svenstaro, 2022-12-09:
Can't this issue be closed as the original request has been fulfilled?

sojusnik, 2022-12-09:
I think this issue is still open due to:

> The current implementation uses systemd for restoring the values, unfortunately flatpak containers don't have one. A flatpak compatible solution wil be in the v0.5.0.

soyersoyer, 2022-12-09:
Yes, it can. It is clearer if we have another issue for the flatpak problem.

closed 2022-12-09 by soyersoyer