soyersoyer/cameractrls #3: Package GTK app as Flatpak

cassidyjames, 2022-06-20:
I'd love to use this app, but my work computer is Endless OS and my personal is Fedora Silverblue. As such, apps are typically just installed via Flatpak. Packaging cameractrls as a Flatpak (and ideally distributing via Flathub) would widen its reach to anyone using a modern Linux-based distro, including the image-based ones like Endless OS, Fedora Silverblue, and even Steam OS.

soyersoyer, 2022-06-21:
Interesting, not as bad as it first seems. I'm in. Stay tuned! :)

soyersoyer, 2022-06-23:
Could you help me?
I couldn't find a solution for using systemd paths and services, and that is necessary for restoring the control values when you connect the camera/start the session.
There is an issue in flatpak
https://github.com/flatpak/flatpak/issues/2787

soyersoyer, 2022-06-23:
One possible solution:
- enable webcam access witih --device=all
- map the .config/systemd with --filesystem=xdg-config/systemd
- dbus access to systemd with --socket=system-bus or --socket=session-bus
- write the service file with ExecStart=flatpak run {appid} --command=cameractrls.py -d {device} -c {controls}
- enable the paths/services with dbus systemd

but this seems a bit too much, and the app needs to know whether it's running inside a flatpak or not, so not too ideal.

sonnyp, 2022-06-27:
> the app needs to know whether it's running inside a flatpak or not

FYI - this is pretty common and not something that's discouraged - libportal even has APIs for it

https://libportal.org/class.Portal.html#type-functions

* `xdp_portal_running_under_flatpak`
* `xdp_portal_running_under_sandbox`
* `xdp_portal_running_under_snap`

liquidat, 2022-07-04:
Thanks for providing the flatpak - there is discovered this app, and it is awesome!

soyersoyer, 2022-07-04:
I'm glad you like it

cassidyjames, 2022-07-27:
I think this can be closed since it's up on Flathub. Thanks for the work, I can now use this for work on Endless OS and Fedora Silverblue!

closed 2022-07-27 by cassidyjames