soyersoyer/cameractrls #4: HDR setting doesn't seem to do anything on a Razer Kiyo Pro

jpetazzo, 2022-09-19:
Hi!

First of all, thanks a lot for writing this tool. I was using `guvcview` before, and I like to have your tool available as an alternative 👍🏻

I'm experiencing weird quality issues on a brand new Razer Kiyo Pro (specifically: even on 60fps the video feels choppy; it looks like maybe the exposure time is too high, no matter what settings I use). That's how I found your tool, with the proprietary Razer controls.

With cameractrls, I can change the FOV of my Razer Kiyo Pro (using the Narrow/Medium/Wide toggles) however when I toggle HDR on or off I don't see any difference. Same if I change HDR to bright or dark.

I wonder if that's normal; i.e. do you typically see a change when you toggle HDR, or maybe it's only visible in some conditions? Or maybe I need to restart the capture each time I change it? Or it works only in some resolutions/modes? (I tried a few combinations but nothing showed a change so far!)

Thank you!

soyersoyer, 2022-09-19:
Hi!
HDR is useful when recording in direct sunlight or if you have very bright spots on the picture.
You can see an example here:
https://www.youtube.com/watch?v=2BT-7UA-6G8&t=633
It reduces the framerate too. I don't think it can be enabled while capturing.
Also try upgrading the firmware, this may fix some issues. Try to use the USB3 port, it might fix some other issues.
You are welcome!

jpetazzo, 2022-09-28:
Hey, thanks for your reply!

I tried to point the camera straight at a bright window, and then I was able to see a difference between HDR on and off, yes!

(The video review that you posted is 100% on point; the presenter said something like "unless you have a very bright window right behind you and can't do anything about it, HDR won't be very useful". I definitely agree, since that's the only scenario where I was able to see a difference; pointing bright lights at the camera didn't change anything visible 😅)

And indeed, HDR can't be enabled/disabled while capturing.
In fact, it needs to be toggled while NOT capturing, i.e. if I do:
- disable HDR
- capture
- while capturing, enable HDR
- stop capture
- restart capture
... Then HDR is still off (even though it shows as "on" in the control panel).

Also, when starting the control panel, HDR shows up as "undetermined" (i.e. neither "on" nor "off" are pressed), is that normal?

soyersoyer, 2022-09-28:
The Kiyo Pro doesn't have a getter function for that settings. I couldn't find it. Razer Synapse in windows saves the state to disk, than shows the last state when, you open it. So it's not normal, it's just limited by the hardware.

jpetazzo, 2022-09-29:
OK, that makes sense! Anyway thanks again for that program; super useful 👍🏻

closed 2022-09-29 by jpetazzo