soyersoyer/cameractrls #5: Gamma slider is locked?

berndhalbherr, 2022-09-30:
I recently installed the app on Linux Mint on an old imac 27, but the gamma got stuck between 100 and 120 (at the start it was 200) now the camera has bad quality, too dark, how can I change it back? Also uninstalling doesn't change anything, and with other apps I cannot access the gamma? Could you recommend a solution?

soyersoyer, 2022-09-30:
That's weird.
Did you change the exposure or enable any auto control?

With v4l2-ctl (from v4l-utils) you can query and set the control values:
```v4l2-ctl -l```
```v4l2-ctl -c gamma=200```

Or you can try qv4l2, v4l2ucp.

Could you insert the output of the ```v4l2-ctl -l```? Thanks!

berndhalbherr, 2022-10-06:
Thank you for your reply,it is OK again, somehow it recovered ^^Actually, I  don’t know how…Best regardsBernd Halbherr Von: soyerGesendet: Freitag, 30. September 2022 22:15An: soyersoyer/cameractrlsCc: berndhalbherr; AuthorBetreff: Re: [soyersoyer/cameractrls] Gamma slider is locked? (Issue #5) That's weird.Did you change the exposure or enable any auto control?With v4l2-ctl (from v4l-utils) you can query and set the control values:v4l2-ctl -lv4l2-ctl -c gamma=200Or you can try qv4l2, v4l2ucp.Could you insert the output of the v4l2-ctl -l? Thanks!

closed 2022-10-06 by soyersoyer