soyersoyer/cameractrls #6: preview is just a black screen

dln949, 2022-12-02:
Not sure what information you need for this.  

Running flatpak of cameractrls version 0.4.11 on Ubuntu 18.04.  Webcam is a Dynex 1.3 MP.  

Clicking on the preview button shows a black screen.  (But the Cheese application shows the video with no problem, so the camera does work.)  

After opening the preview window (which, as I wrote above, is simply a black screen), I can't close it, I need to do a "force quit" because I get the message that cameractrls is not responding.  

I tried running flatpak run hu.irl.cameractrls in the terminal with the hope that I would see some messages, but no messages were displayed at all.

soyersoyer, 2022-12-02:
Could you send the output of the ```flatpak run --command=cameractrls.py hu.irl.cameractrls -l```?
Could you try it with different pixel format or resolution? You can change it on Capture tab.

The cameraview.py (the preview window) is very simple at the moment, there is only one thread. If the camera blocks it for a long time, the event handlers cannot run, so the window cannot be closed. I'll fix it soon.

dln949, 2022-12-02:
Changing the pixels and resolution did not make any discernable difference.  

Here is the terminal output you asked for (I can't figure out how to paste in the terminal output to avoid all that strikeout text, don't know why that is being added by Github):  

main1@system1:~$ flatpak run --command=cameractrls.py hu.irl.cameractrls -l
Basic / Crop
 zoom_absolute = 0	( default: 0 min: 0 max: 3 )
 pan_absolute = 0	( default: 0 min: -57600 max: 57600 step: 3600 )
 tilt_absolute = 0	( default: 0 min: -43200 max: 43200 step: 3600 )
Exposure / Exposure
 exposure_auto = aperture_priority_mode	( default: aperture_priority_mode values: manual_mode, aperture_priority_mode ) | updater
 exposure_absolute = 332	( default: 300 min: 1 max: 5000 ) | inactive
 exposure_auto_priority = 1	( default: 0 min: 0 max: 1 )
 gain = 0	( default: 0 min: 0 max: 100 )
Exposure / Dynamic Range
 backlight_compensation = 1	( default: 1 min: 0 max: 2 )
Color / Balance
 white_balance_temperature_auto = 1	( default: 1 min: 0 max: 1 ) | updater
 white_balance_temperature = 6500	( default: 6500 min: 2800 max: 6500 ) | inactive
Color / Color
 brightness = 0	( default: 0 min: -64 max: 64 )
 contrast = 32	( default: 32 min: 0 max: 64 )
 saturation = 64	( default: 64 min: 0 max: 128 )
 sharpness = 32	( default: 32 min: 0 max: 64 )
 hue = 0	( default: 0 min: -40 max: 40 )
 gamma = 100	( default: 100 min: 72 max: 500 )
Advanced / Power Line
 power_line_frequency = 60_hz	( default: 60_hz values: disabled, 50_hz, 60_hz )
Capture / Capture
 pixelformat = YUYV	( values: YUYV, MJPG )
 resolution = 1280x1024	( values: 640x480, 352x288, 320x240, 176x144, 160x120, 1280x1024 )
 fps = 9	( values: 9, 5 )
main1@system1:~$

soyersoyer, 2022-12-02:
Do you see any usb related error at the end of your dmesg?

dln949, 2022-12-02:
Don't know what any of this means when I run dmesg, I hope you do:  

[16221.671819] usb 3-13.3.3: new high-speed USB device number 21 using xhci_hcd
[16221.822701] usb 3-13.3.3: New USB device found, idVendor=19ff, idProduct=0102, bcdDevice= 1.00
[16221.822703] usb 3-13.3.3: New USB device strings: Mfr=2, Product=1, SerialNumber=0
[16221.822705] usb 3-13.3.3: Product: Dynex 1.3MP Webcam
[16221.822706] usb 3-13.3.3: Manufacturer: Dynex
[16221.822906] usb 3-13.3.3: Device is not authorized for usage
[16221.829932] uvcvideo: Found UVC 1.00 device Dynex 1.3MP Webcam (19ff:0102)
[16221.853948] uvcvideo 3-13.3.3:1.0: Entity type for entity Extension 5 was not initialized!
[16221.853951] uvcvideo 3-13.3.3:1.0: Entity type for entity Extension 4 was not initialized!
[16221.853952] uvcvideo 3-13.3.3:1.0: Entity type for entity Processing 3 was not initialized!
[16221.853954] uvcvideo 3-13.3.3:1.0: Entity type for entity Camera 1 was not initialized!
[16221.854142] input: Dynex 1.3MP Webcam: Dynex 1.3MP as /devices/pci0000:00/0000:00:14.0/usb3/3-13/3-13.3/3-13.3.3/3-13.3.3:1.0/input/input37
[16221.883846] usb 3-13.3.3: 3:1: cannot get freq at ep 0x84
[16221.893392] usb 3-13.3.3: Warning! Unlikely big volume range (=3328), cval->res is probably wrong.
[16221.893397] usb 3-13.3.3: [2] FU [Mic Capture Volume] ch = 1, val = 4608/7936/1
[16221.894521] usb 3-13.3.3: authorized to connect
[16222.013462] usb 3-13.3.3: 3:1: cannot get freq at ep 0x84
[16222.049334] usb 3-13.3.3: 3:1: cannot get freq at ep 0x84

soyersoyer, 2022-12-02:
I have pushed some new changes.
Could you try it with the main branch?
```
sudo apt install libsdl2-2.0-0 libturbojpeg git
git clone https://github.com/soyersoyer/cameractrls.git
cd cameractrls
./cameractrlsgtk.py
```
After opening the preview window, could you please post the output of the ```./cameractrlsgtk.py``` along with the end of the ```dmesg```?
Thanks!

dln949, 2022-12-03:
Running the py program the preview window did work, but the main window was just a white box, the only control on it was the one to view the preview window.  

main1@system1:~/cameractrls$ ./cameractrlsgtk.py
Traceback (most recent call last):
  File "./cameractrlsgtk.py", line 69, in <lambda>
    self.device_cb.connect('changed', lambda e: self.gui_open_device(self.device_cb.get_active()))
  File "./cameractrlsgtk.py", line 120, in gui_open_device
    self.init_gui_device()
  File "./cameractrlsgtk.py", line 289, in init_gui_device
    wb_cb.set_active(find_idx(c.menu, lambda m: m.text_id == c.value))
TypeError: Argument 1 does not allow None as a value
WARNING:root:/dev/v4l/by-id/usb-Dynex_Dynex_1.3MP_Webcam-video-index0: timeout occured
main1@system1:~/cameractrls$ 

dmesg:  

[38237.687003] usb 3-13.3.3: new high-speed USB device number 22 using xhci_hcd
[38237.833902] usb 3-13.3.3: New USB device found, idVendor=19ff, idProduct=0102, bcdDevice= 1.00
[38237.833904] usb 3-13.3.3: New USB device strings: Mfr=2, Product=1, SerialNumber=0
[38237.833905] usb 3-13.3.3: Product: Dynex 1.3MP Webcam
[38237.833906] usb 3-13.3.3: Manufacturer: Dynex
[38237.834066] usb 3-13.3.3: Device is not authorized for usage
[38237.841308] uvcvideo: Found UVC 1.00 device Dynex 1.3MP Webcam (19ff:0102)
[38237.865193] uvcvideo 3-13.3.3:1.0: Entity type for entity Extension 5 was not initialized!
[38237.865194] uvcvideo 3-13.3.3:1.0: Entity type for entity Extension 4 was not initialized!
[38237.865195] uvcvideo 3-13.3.3:1.0: Entity type for entity Processing 3 was not initialized!
[38237.865197] uvcvideo 3-13.3.3:1.0: Entity type for entity Camera 1 was not initialized!
[38237.865300] input: Dynex 1.3MP Webcam: Dynex 1.3MP as /devices/pci0000:00/0000:00:14.0/usb3/3-13/3-13.3/3-13.3.3/3-13.3.3:1.0/input/input41
[38237.894842] usb 3-13.3.3: 3:1: cannot get freq at ep 0x84
[38237.904378] usb 3-13.3.3: Warning! Unlikely big volume range (=3328), cval->res is probably wrong.
[38237.904380] usb 3-13.3.3: [2] FU [Mic Capture Volume] ch = 1, val = 4608/7936/1
[38237.904754] usb 3-13.3.3: authorized to connect
[38238.033231] usb 3-13.3.3: 3:1: cannot get freq at ep 0x84
[38238.073602] usb 3-13.3.3: 3:1: cannot get freq at ep 0x84
main1@system1:~$

soyersoyer, 2022-12-03:
That's strange.
Please run ```git pull``` and re run the ```cameractrlsgtk.py``` and post the output.
Could you post the output of the ```v4l2-ctl --list-formats-ext``` too?

dln949, 2022-12-03:
git pull   does not work:  

main1@system1:~$ git pull
fatal: not a git repository (or any of the parent directories): .git
main1@system1:~$

soyersoyer, 2022-12-03:
Git commands usually works on the working directory, just change it to the cameractrls `cd cameractrls`.

You can quote your posts with [backticks](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code)

dln949, 2022-12-03:
main1@system1:~$ cd cameractrls
main1@system1:~/cameractrls$ git pull
remote: Enumerating objects: 4, done.
remote: Counting objects: 100% (4/4), done.
remote: Total 4 (delta 3), reused 4 (delta 3), pack-reused 0
Unpacking objects: 100% (4/4), done.
From https://github.com/soyersoyer/cameractrls
   26dfa6a..f82ca4a  main       -> origin/main
Updating 26dfa6a..f82ca4a
Fast-forward
 cameractrls.py    | 3 ++-
 cameractrlsgtk.py | 8 ++++++--
 2 files changed, 8 insertions(+), 3 deletions(-)
main1@system1:~/cameractrls$ cameractrlsgtk.py
cameractrlsgtk.py: command not found
main1@system1:~/cameractrls$ v4l2-ctl --list-formats-ext
Failed to open /dev/video0: No such file or directory
main1@system1:~/cameractrls$

soyersoyer, 2022-12-04:
Great! Thanks for testing. These changes will be included in the upcoming version v0.4.12.
Your fps control still bothers me. Could you post the output of the `v4l2-ctl --all` after a fresh usb connection?

dln949, 2022-12-04:
When you say "upcoming version v0.4.12", do you mean that will be in the flatpak version?  

The output you asked for:  

```
main1@system1:~/cameractrls$ v4l2-ctl --all
Driver Info (not using libv4l2):
	Driver name   : uvcvideo
	Card type     : Dynex 1.3MP Webcam: Dynex 1.3MP
	Bus info      : usb-0000:00:14.0-13.3.3
	Driver version: 4.18.20
	Capabilities  : 0x84A00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 1280/1024
	Pixel Format      : 'YUYV'
	Field             : None
	Bytes per Line    : 2560
	Size Image        : 2621440
	Colorspace        : Default
	Transfer Function : Default (maps to Rec. 709)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Limited Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 1280, Height 1024
	Default     : Left 0, Top 0, Width 1280, Height 1024
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 1280, Height 1024
Selection: crop_bounds, Left 0, Top 0, Width 1280, Height 1024
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 9.000 (9/1)
	Read buffers     : 0
                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=-13
                       contrast 0x00980901 (int)    : min=0 max=64 step=1 default=32 value=43
                     saturation 0x00980902 (int)    : min=0 max=128 step=1 default=64 value=85
                            hue 0x00980903 (int)    : min=-40 max=40 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=72 max=500 step=1 default=100 value=100
                           gain 0x00980913 (int)    : min=0 max=100 step=1 default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=6500 value=6500 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=64 step=1 default=32 value=32
         backlight_compensation 0x0098091c (int)    : min=0 max=2 step=1 default=1 value=1
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
              exposure_absolute 0x009a0902 (int)    : min=1 max=5000 step=1 default=300 value=332 flags=inactive
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=1
                   pan_absolute 0x009a0908 (int)    : min=-57600 max=57600 step=3600 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=-43200 max=43200 step=3600 default=0 value=0
                  zoom_absolute 0x009a090d (int)    : min=0 max=3 step=1 default=0 value=0
main1@system1:~/cameractrls$ ```

soyersoyer, 2022-12-04:
Thanks. The `Frames per second: 9.000 (9/1)` looks normal.

Yes, that will be here in the v0.4.12 tag first, then in the Flathub/Arch/Snap versions.

dln949, 2022-12-05:
This morning I saw that I received an update to the flatpak package of cameractrls, it is now at version 0.4.12.  However, it still has the same two problems that I first described in this issue.

soyersoyer, 2022-12-05:
Could you check version is 0.4.12 also in the About? You can open it from the top right hamburger menu.

dln949, 2022-12-06:
That is where I got the version number from.

soyersoyer, 2022-12-06:
That's interesting. If you run it from the terminal do you see any messages? What is your flatpak version? The flatpak [setup page](https://flatpak.org/setup/Ubuntu) recommends their own PPA for 18.04.

dln949, 2022-12-06:
```
main1@system1:~$ flatpak --version
Flatpak 1.12.4
main1@system1:~$ 
```

```
main1@system1:~$ flatpak run hu.irl.cameractrls
Gtk-Message: 08:05:52.906: Failed to load module "canberra-gtk-module"

```

soyersoyer, 2022-12-06:
So it works when you run it in 18.04, but inside flatpak the preview window (SDL) is just black and unresponsive? Does the setting part (GTK) work?

Maybe something with Flatpak or SDL.

Could you run it with SDL logging? `SDL_EVENT_LOGGING=1 flatpak run hu.irl.cameractrls`

dln949, 2022-12-06:
Correct, it works fine running your python script, but the flatpak  package has the two problems.  

The settings part (GTK) does work even when run from the flatpak.  

```
main1@system1:~$ SDL_EVENT_LOGGING=3 flatpak run hu.irl.cameractrls
Gtk-Message: 17:25:24.620: Failed to load module "canberra-gtk-module"
INFO: SDL EVENT: SDL_WINDOWEVENT (timestamp=111 windowid=1 event=SDL_WINDOWEVENT_SHOWN data1=0 data2=0)
INFO: SDL EVENT: SDL_WINDOWEVENT (timestamp=111 windowid=1 event=SDL_WINDOWEVENT_HIDDEN data1=0 data2=0)
INFO: SDL EVENT: SDL_WINDOWEVENT (timestamp=239 windowid=1 event=SDL_WINDOWEVENT_SHOWN data1=0 data2=0)
INFO: SDL EVENT: SDL_WINDOWEVENT (timestamp=267 windowid=1 event=SDL_WINDOWEVENT_MOVED data1=2240 data2=71)
INFO: SDL EVENT: SDL_WINDOWEVENT (timestamp=267 windowid=1 event=SDL_WINDOWEVENT_MOVED data1=2240 data2=58)
INFO: SDL EVENT: SDL_WINDOWEVENT (timestamp=267 windowid=1 event=SDL_WINDOWEVENT_MOVED data1=2240 data2=71)
INFO: SDL EVENT: SDL_WINDOWEVENT (timestamp=268 windowid=1 event=SDL_WINDOWEVENT_EXPOSED data1=0 data2=0)
INFO: SDL EVENT: SDL_WINDOWEVENT (timestamp=268 windowid=1 event=SDL_WINDOWEVENT_FOCUS_GAINED data1=0 data2=0)
INFO: SDL EVENT: SDL_WINDOWEVENT (timestamp=268 windowid=1 event=SDL_WINDOWEVENT_TAKE_FOCUS data1=0 data2=0)
WARNING:root:/dev/v4l/by-id/usb-Dynex_Dynex_1.3MP_Webcam-video-index0: timeout occured
INFO: SDL EVENT: SDL_WINDOWEVENT (timestamp=1367 windowid=1 event=SDL_WINDOWEVENT_ENTER data1=0 data2=0)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1367 windowid=1 which=0 state=0 x=51 y=314 xrel=0 yrel=0)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1375 windowid=1 which=0 state=0 x=114 y=313 xrel=63 yrel=-1)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1383 windowid=1 which=0 state=0 x=184 y=312 xrel=70 yrel=-1)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1391 windowid=1 which=0 state=0 x=248 y=307 xrel=64 yrel=-5)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1399 windowid=1 which=0 state=0 x=310 y=304 xrel=62 yrel=-3)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1407 windowid=1 which=0 state=0 x=363 y=301 xrel=53 yrel=-3)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1415 windowid=1 which=0 state=0 x=423 y=299 xrel=60 yrel=-2)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1423 windowid=1 which=0 state=0 x=465 y=296 xrel=42 yrel=-3)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1431 windowid=1 which=0 state=0 x=498 y=293 xrel=33 yrel=-3)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1439 windowid=1 which=0 state=0 x=533 y=287 xrel=35 yrel=-6)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1447 windowid=1 which=0 state=0 x=554 y=284 xrel=21 yrel=-3)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1455 windowid=1 which=0 state=0 x=567 y=281 xrel=13 yrel=-3)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1463 windowid=1 which=0 state=0 x=579 y=278 xrel=12 yrel=-3)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1471 windowid=1 which=0 state=0 x=589 y=274 xrel=10 yrel=-4)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1479 windowid=1 which=0 state=0 x=596 y=271 xrel=7 yrel=-3)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1487 windowid=1 which=0 state=0 x=600 y=268 xrel=4 yrel=-3)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1495 windowid=1 which=0 state=0 x=606 y=264 xrel=6 yrel=-4)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1503 windowid=1 which=0 state=0 x=613 y=260 xrel=7 yrel=-4)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1511 windowid=1 which=0 state=0 x=623 y=254 xrel=10 yrel=-6)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1519 windowid=1 which=0 state=0 x=635 y=250 xrel=12 yrel=-4)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1527 windowid=1 which=0 state=0 x=646 y=244 xrel=11 yrel=-6)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1535 windowid=1 which=0 state=0 x=661 y=238 xrel=15 yrel=-6)
INFO: SDL EVENT: SDL_MOUSEMOTION (timestamp=1543 windowid=1 which=0 state=0 x=671 y=234 xrel=10 yrel=-4)
INFO: SDL EVENT: SDL_USEREVENT (timestamp=1547 windowid=0 code=0 data1=0x7fed0c0f2000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=1654 windowid=0 code=0 data1=0x7fed0be72000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=1766 windowid=0 code=0 data1=0x7fed0bbf2000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=1879 windowid=0 code=0 data1=0x7fed0b972000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=1987 windowid=0 code=0 data1=0x7fed0b6f2000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=2098 windowid=0 code=0 data1=0x7fed0b472000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=2206 windowid=0 code=0 data1=0x7fed0c0f2000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=2318 windowid=0 code=0 data1=0x7fed0be72000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=2426 windowid=0 code=0 data1=0x7fed0bbf2000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=2538 windowid=0 code=0 data1=0x7fed0b972000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=2646 windowid=0 code=0 data1=0x7fed0b6f2000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=2758 windowid=0 code=0 data1=0x7fed0b472000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=2871 windowid=0 code=0 data1=0x7fed0c0f2000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=3018 windowid=0 code=0 data1=0x7fed0be72000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=3186 windowid=0 code=0 data1=0x7fed0bbf2000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=3366 windowid=0 code=0 data1=0x7fed0b972000 data2=(nil))
[... cut ...]
INFO: SDL EVENT: SDL_USEREVENT (timestamp=14350 windowid=0 code=0 data1=0x7fed0b6f2000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=14550 windowid=0 code=0 data1=0x7fed0b472000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=14750 windowid=0 code=0 data1=0x7fed0c0f2000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=14951 windowid=0 code=0 data1=0x7fed0be72000 data2=(nil))
INFO: SDL EVENT: SDL_USEREVENT (timestamp=15150 windowid=0 code=0 data1=0x7fed0bbf2000 data2=(nil))
main1@system1:~$ 
```

dln949, 2022-12-07:
No selinux, I do have some apparmor profiles - that I did not manually create - none of the refer to camerctrls.  

Graphics card info:  Intel® HD Graphics 4600 (HSW GT2)

terminal output:  

```
main1@system1:~$ cd cameractrls.py
bash: cd: cameractrls.py: No such file or directory
main1@system1:~/cameractrls$ LIBGL_ALWAYS_SOFTWARE=1 flatpak run hu.irl.cameractrls or flatpak --device=dri run hu.irl.cameractrls
Gtk-Message: 20:16:41.488: Failed to load module "canberra-gtk-module"
WARNING:root:/dev/v4l/by-id/usb-Dynex_Dynex_1.3MP_Webcam-video-index0: timeout occured

(process:2000): flatpak-WARNING **: 20:17:00.388: Error writing credentials to socket: Error sending message: Broken pipe
main1@system1:~/cameractrls$ 
```

Running the above command cameractrls worked perfectly.

soyersoyer, 2022-12-07:
Glad to read.
You can override LIBGL_ALWAYS_SOFTWARE env variable inside flatpak with:
`flatpak override --user --env=LIBGL_ALWAYS_SOFTWARE=1 hu.irl.cameractrls`
or system wide:
`sudo flatpak override --env=LIBGL_ALWAYS_SOFTWARE=1 hu.irl.cameractrls`

dln949, 2022-12-07:
Before I do that, I need to understand what doing that will mean, what it will affect.  

I don't want to do something to make cameractrls' flatpak package work correctly only to find that I've caused a problem for all other installed flatpak packages on my system.  

Or, is what you're suggesting going to affect ONLY cameractrls?  

Either way - can you briefly describe what the problem is that your solution is addressing, and how it will fix that problem?  

Thanks.

soyersoyer, 2022-12-07:
It only affects cameractrls.

```
$ flatpak override --help
Usage:
  flatpak override [OPTION…] [APP] - Override settings [for application]
```

I think the Mesa drivers are different in flatpak and your system, they can't work together well. With the LIBGL_ALWAYS_SOFTWARE=1 SDL uses a software renderer., which is slower. It only affects the preview window.

dln949, 2022-12-07:
Thanks for the explanation.  I tried it, it works great.  Thanks for all your help.

closed 2022-12-07 by dln949