soyersoyer/cameractrls #7: Pure black icon color is invisible on black panels

nekohayo, 2022-12-03:
If you use [dash-to-panel](https://extensions.gnome.org/extension/1160/dash-to-panel/), or some other launcher, or other desktop environments than a vanilla GNOME, the icon's colors will prove problematic, as only the small red dot remains visible:

[image]

Generally speaking, pure black does not exist in nature, so it might look more natural if the icon uses some shade of gray, white, subtle gradients or textures.

eldelacajita, 2022-12-04:
@soyersoyer would you be interested in a new icon that solves this and aligns better with the GNOME/Adwaita style?

I would love to give it a try, no obligation from your side.

soyersoyer, 2022-12-04:
@nekohayo Good point, thanks for your suggestion.
@eldelacajita Yes, that would be cool :) Please try to preserve the red dot and the simplicity.

soyersoyer, 2022-12-04:
I added a bloom effect to the current icon, but I still want a GNOME/Adwaita style icon. :)

eldelacajita, 2022-12-05:
I'm on it!

Here are some first attempts, in two different directions and with slightly different styles.

## Physical camera and sliders representing "settings"
I have used simple shapes, no reflections, no gradients, but some drop shadows.

[image]

[image]

## A different approach, combining the settings icon with a camera
Here I tried a slightly more detailed style, with some gradients and reflections.

[image]

What do you think? Feel free to tell me you want something entirely different, I'm having fun exploring anyway :smile:

soyersoyer, 2022-12-09:
A not so black icon in v0.4.12. The new with GNOME style is being made. Thanks.

closed 2022-12-09 by soyersoyer