soyersoyer/cameractrls #8: New icon

eldelacajita, 2022-12-05:
Creating this as a separate issue, coming from the conversation at #7.

Here are some first attempts, in two different directions and with slightly different styles.

## Physical camera and sliders representing "settings"
I have used simple shapes, no reflections, no gradients, but some drop shadows.

[image]

This variation ended up looking more like an action camera, but I'm including it anyway.

[image]

## A different approach, combining the settings icon with a camera
Here I tried a slightly more detailed style, with some gradients and reflections.

[image]

What do you think? Feel free to tell me you want something entirely different, I'm having fun exploring anyway :smile:

eldelacajita, 2022-12-05:
Here I come with some more variations: 

## Separating camera and settings 
A combination of a more webcam-like camera and a settings panel that appears under it. 

[image]

The same, but moving the red dot and adding a microphone hole to make it somehow symmetric. Also, slightly adjusting the width of the settings panel to make it look more balanced (in the version above, it looks wider than the camera because it's brighter).

[image]

eldelacajita, 2022-12-05:
Once we settle on one of these concepts, I can polish it more and make the symbolic icon, too.

soyersoyer, 2022-12-05:
Wow, these are cool! I like the 05c best.
Could you add 2 or 3 more mic hole for future safety?
I think the lens could be a bit bigger, like sony compacts (eg a6000, zv1) where the lens is almost as tall as the body.

eldelacajita, 2022-12-07:
Those are great references, thank you!

I'll try to find a balance between keeping it simple and generic, but also evocative of those devices.

eldelacajita, 2022-12-08:
Here's another variant: 

[image]

And the same with a more detailed style: 

[image]

I'm not sure if I like it better than the 5c, though. To make the lens bigger and still have space for two mic holes each side, I had to make the camera less tall and give more proportional space to the settings panel. Also, it becomes less abstract, and we could end up sliding into a rabbit hole trying to imitate real life cameras.

Anyway, let me know what you think!

soyersoyer, 2022-12-08:
I like the 06b. It feels more modern to me. Maybe because of the camera's proportions. Could you make one more version of the 06b without the double mic, and another version of the 05c with detailed lens?

eldelacajita, 2022-12-10:
Here they are. I also noticed that I was making the 5c variant too tall for the template, so after adjusting that too, they both ended being quite similar except for the size of the lens: 

[image]

[image]

I think any of these last variants (6b, 6b-2, 5c-2) can work, so we can choose the one you like the most, with some final touches if needed.

eldelacajita, 2023-02-19:
Hi, @soyersoyer! How is everything going? Let me know if you want me to continue with this and export the final icon. Which one do you prefer?

soyersoyer, 2023-03-02:
Sorry for the late reply.
Could we have one more version?
My idea is to
color the lens reflection from white to red(ish),
highlight the aperture blades (5-6), to notice this is a settings app.
so we can remove the light part with the slider, the red light, the microphone, and possibly the camera body as well.
What do you think?

eldelacajita, 2023-03-05:
Something like this?

[image]

soyersoyer, 2023-04-26:
Yes, it's very good! If you export it, I can change the icon to it.

eldelacajita, 2023-05-14:
Hi! Sorry for the delay. Here it is:

Final version with included symbolic icon:

[image]

Optimized version, ready to use (exported from the Icon preview app)

[image]

nekohayo, 2023-05-14:
There's one issue with the closed aperture lens being used as a standalone icon without the "camera body" though, if you intend the details / blades to be visible, they don't really seem to be; not enough contrast between the dark gray of the blades and the pure black of the iris and space between blades, I think. This will be particularly hard to interpret visually on a light background (when gnome shell eventually gets a light theme, maybe) or on uncalibrated monitors.

In reality, unless the scene is completely dark, aperture blades are usually much brighter, and the division lines between the blades have shades and highlights that depend on the angle the light is at, they're not pure black everywhere (you could probably use a bunch of gradients for this). Blades themselves will tend to have a gradient instead of being totally flat, but maybe that's an aesthetic choice here.

See the photos of aperture blades there for reference:

* https://en.wikipedia.org/wiki/Aperture#/media/File:Canon_EF-M_32mm_F1.4_STM_lens-aperture_blades_PNr%C2%B00803.jpg
* https://photographylife.com/wp-content/uploads/2022/10/ApertureBlades.jpg
* https://petapixel.com/assets/uploads/2022/09/aperture-in-photography-featured-image.jpg (there are also many other different aperture blade photos in [that article](https://petapixel.com/what-is-aperture-in-photography/) that show the looks depending on the ambient lighting)

nekohayo, 2023-05-14:
That said, this is less of an issue if the lens is embedded in the camera "body" with its sliders, because then it's not "just a circle like the gazillion other camera apps that are just a lens"; in that case it can make sense for the aperture blades to be a bit less prominent.

I do think the versions embedded in a camera body with sliders are more unique, recognizeable, and "understandable in what the app's function is" than just the lens.

nekohayo, 2023-05-14:
...and one more idea: if you want to give the icon a truly distinctive shape/silouhette (instead of being yet another circle / rectangle), you could grab one of those "webcam sitting atop a settings panel" variants above and trim the width of the settings area (the white part, which has thick margins on the left and right anyway) so that it's not as wide as the camera's gray body area, and this would give it a "T" shape that is very much like the shape of actual webcams with their foot. Essentially, you'd be hiding the settings sliders into a white foot ;)

eldelacajita, 2023-05-14:
> In reality, unless the scene is completely dark, aperture blades are usually much brighter, and the division lines between the blades have shades and highlights that depend on the angle the light is [...]

Yes, it's all very simplified by choice. We are not aiming at a realistic/skeuomorphic design.

That said, you have a point here:

> if you intend the details / blades to be visible, they don't really seem to be

I could try to improve that, although I'd probably have to change several colors to make this work better within the restricted GNOME palette.

eldelacajita, 2023-05-14:
> I do think the versions embedded in a camera body with sliders are more unique, recognizeable, and "understandable in what the app's function is" than just the lens.

I agree. But regarding that, and your other suggestion, I'll wait to see if @soyersoyer wants to go that way.

eldelacajita, 2023-05-14:
> I could try to improve that, although I'd probably have to change several colors to make this work better within the restricted GNOME palette.

[image]

I tried making the blade lines a bit thicker, the blades a bit lighter, and the outer ring darker. I'm not very convinced of this. One challenge is to keep enough contrast between the blades and the red reflection (I want a dark background for it to stand out). Another challenge is to keep this looking more or less right without trying to go realistic.

eldelacajita, 2023-05-16:
Sure, here it is:

[image]

soyersoyer, 2023-05-16:
I like the 7a better. It looks great with the red flare in the apps.
I think it's better than 6b, because 6b can be the logo of any camera control app, but this is more unique even if there are apps with lens logo.
Thanks. I'll add it to the project soon.
And sorry for the slow replies.

eldelacajita, 2023-05-25:
Glad to see the new icon among my apps!

This issue can be closed, I suppose.

soyersoyer, 2023-05-25:
I also added you to the About popup. Thanks for everything.

closed 2023-05-25 by soyersoyer