soyersoyer/cameractrls #9: Confused about Save (Fedora + Flatpak)!

benwilcock, 2022-12-09:
Hey! 

Great app, thank you!

Sorry to bother you, but I'm wondering what the trick is to activate the Systemd "Save" button.

I'm on Fedora 37 and using the Flatpak installation method. I have Flatseal and tried opening up file access, but no luck.

Can you help?

Cheers

Ben

soyersoyer, 2022-12-09:
Sorry, it isn't supported yet. Unfortunately, Flatpak containers can't use Systemd.
I want to create a very small cameractrls daemon in rust to restore the controls, that can be started at startup with [org.freedesktop.portal.Background](https://docs.flatpak.org/en/latest/portal-api-reference.html#gdbus-org.freedesktop.portal.Background).
Stay tuned, it will be in v0.5.

benwilcock, 2022-12-11:
OK. Cool. Thanks for clarifying. Will await v0.5. Thanks again!

hyuri, 2022-12-12:
Hi, I'm confused about the question. Is it about having the ability to save the settings with a button — or automatically, as you change the settings? I came here to ask about that.

soyersoyer, 2023-06-12:
Unfortunately, I haven't finished it yet. Stay tuned :)

soyersoyer, 2024-03-31:
Fixed in 0.6.0. Save your settings into Preset 1, and enable the Cameractrlsd with Desktop Portal (on Advanced page)

closed 2024-03-31 by soyersoyer