soyersoyer/cameractrls #17: "0 camera found"

Serg10, 2023-04-12:
I have a laptop with integrated webcam and a usb cam, both work with browser, telegram, etc... but cameractrls says "0 camera found" 
DELL latitude 7350
Ubuntu 22.04.2 LTS
cameractrls version 0.4.14-0 

[image]

soyersoyer, 2023-04-26:
Try the `snap connect cameractrls:camera` as described at https://snapcraft.io/cameractrls

Serg10, 2023-04-27:
Yes, I was using the snap version, but I uninstalled it and installed it via flatpak and everything works fine!

closed 2023-04-27 by Serg10

asfernandes, 2025-09-01:
I have the same problem with snap and flatpak.
How to add a camera manually?