soyersoyer/cameractrls #18: Built-in preview as a sidebar instead of a foating window

nekohayo, 2023-04-14:
Having the camera preview _only_ available as a big standalone floating window on top of the app, which you have to manage separately etc. and mess with the window manager to resize and move "out of the way" to be able to continue playing with the settings, is a bit a pain in the butt.

Wouldn't it be nicer if clicking the camera preview button could embed the preview within the main window, on the side? To do so, it should calculate the camera's aspect ratio, clamp the height to match the existing window's measured height, and calculate the width accordingly. If not enough horizontal space is available on the screen, it could place the preview sink above or below instead of on the side..

A button could be added to "pop out" that widget from the window if some users really still want a standalone preview window, but I'm not sure why anyone would want that, it's mostly a temporary thing to set up your camera before a videoconferencing (or recording) session, not something you keep open previewing at all times (because then it would lock access to the camera and prevent the actual videoconferencing from working properly)...

soyersoyer, 2023-04-27:
It's a good idea.
The current preview window (SDL) cannot be placed next to the app window in Wayland. One possible solution is to add  the preview as a widget. For example GTK4 has the new GtkVideo, I think it's time to make a GTK4 GUI.

soyersoyer, 2023-05-20:
It turns out that adding all supported camera formats (YUYV, YVYU, UYVY, NV12, NV21, YU12, RGBP, RGB3, BGR3, RX24, MJPG, JPEG, GREY) to GTK4 is not that easy. And it won't be as fast and efficient as in SDL. I need to find out what other solutions are there. (sdl gui, sdl gl context inside gtk, or some wayland magic, gstreamer)
In v0.5.2, the size of the preview image is limited to fit next to the window, but you have to move it manually.

soyersoyer, 2023-05-21:
I limited the size of the preview window more, so the WM can place it next to or below the control window (if there is free space). You can try it in v0.5.4.

soyersoyer, 2023-08-02:
Gnome is moving towards a [mosaic style](https://blogs.gnome.org/tbernard/2023/07/26/rethinking-window-management/) window management. That will arrange the preview window better.