soyersoyer/cameractrls #19: Razer Kiyo Pro Ultra PCAP needed

soyersoyer, 2023-04-27:
The Razer Kiyo Pro Ultra is the new 'best webcam'. It would be cool to support its special features.
If you have one, please capture USB communication with Wireshark while you change special settings in windows with synapse.

pohlt, 2023-06-01:
I should get mine next week and will hopefully find the time to provide the required information...

pohlt, 2023-06-06:
Got my Kiyo Pro Ultra this morning. \o/

"non standard" are all the messages which are not identifed in the info string I guess: "Zoom (absolute)" is recognized, "Unit 6 control 0x1" is not, right?

Do you need:
- [ ] sent SET CUR Request packages
- [ ] anything sent to the device
- [ ] responses (which responses because if I include the image stream is gets BIG)

Should I send it as PM? I can also put it on my Google Drive and share the folder with you. What's most convenient for you?

soyersoyer, 2023-06-06:
Yes, there are standard uvc controls, that wireshark/linux uvc/anything handles well. Other controls can usually be used with [extension units](https://www.kernel.org/doc/html/v4.9/media/v4l-drivers/uvcvideo.html).
Could you filter with `frame.protocols == "usb:usbvideo"`?
Then you can upload them to google drive and post the links here or create a git repo for them.

pohlt, 2023-06-06:
Here's a first try with the _preview resolution_ and the _2d/3d noise reduction_. Please find the captures and the desciption file [here](https://drive.google.com/drive/folders/1vwVWmW0LfPsV0cmpa5G6GOR1KqKbfDW4?usp=sharing).

Once I see how you would add it to your repo, I might be able to do the rest by myself and create a PR.

soyersoyer, 2023-06-06:
Ah, okey :)

So the noise reduction values in changing order:
c00e020000000000
c00e010100000000
c00e020100000000
c00e010100000000
c00e020100000000
c00e010000000000
c00e010100000000
c00e020100000000

I assume
2D_NR_OFF = c00e020000000000
2D_NR_ON = c00e020100000000
3D_NR_OFF = c00e010000000000
3D_NR_ON = c00e010100000000

I only need an extension unit guid to reliably get the bUnitID.
This can get from  `lsusb -v -d 1532:0E08|grep -B 2 guidExtensionCode`

The preview resolution change should work without any modification. It shouldn't use these control things.

pohlt, 2023-06-06:
```
❯ lsusb -v -d 1532:0E08 | grep -B 2 guidExtensionCode
Couldn't open device, some information will be missing
        bDescriptorSubtype      6 (EXTENSION_UNIT)
        bUnitID                 2
        guidExtensionCode         {2c49d16a-32b8-4485-3ea8-643a152362f2}
--
        bDescriptorSubtype      6 (EXTENSION_UNIT)
        bUnitID                 6
        guidExtensionCode         {23e49ed0-1178-4f31-ae52-d2fb8a8d3b48}
```

soyersoyer, 2023-06-06:
Same as Kiyo  Pro.
I've created a [branch](https://github.com/soyersoyer/cameractrls/tree/kiyoproultra) 
Could you try it?

pohlt, 2023-06-07:
Works! I can't tell the difference for 2D NR on/off (it's the same with Synapse), but turning 3D NR on/off is very obvious. Awesome! Thanks a lot!

I'll let you know when I have more captures.

pohlt, 2023-06-07:
If exposure is manual, you can set the ISO value:
* ISO 100: c009010100000000
* ISO 200: c009010200000000
* ISO 400: c009010300000000
* ISO 800: c009010400000000
* ISO 1600: c009010500000000
* ISO 3200: c009010600000000
* ISO 6400: c009010700000000

pohlt, 2023-06-07:
# Focus  - Auto
### Mode
* Standard: c00a010000000000
* Face: c00a010100000000

### Tracking:
* Passive: ff06010000000000
* Responsive:  ff06000000000000

### Lighting Type
* Standard: c00a010000000000
* Stylized: c00a010000000001

# Image
### Mirror Video
* on:  c00e030100000000
* off: c00e030000000000

pohlt, 2023-06-08:
# Lens Distortion
* off: ff01000300000000
* on: ff01010300000000

soyersoyer, 2023-06-08:
Shutter speed is interesting, there is UVC control for that.  It doesn't use that?
Could you post the output of the `./cameractrls.py  -l`?

soyersoyer, 2023-06-08:
I've added some new controls to [kiyoproultra](https://github.com/soyersoyer/cameractrls/tree/kiyoproultra).

pohlt, 2023-06-09:
> Shutter speed is interesting, there is UVC control for that. Doesn't it use the V4L2_CID_EXPOSURE_ABSOLUTE (CT_EXPOSURE_TIME_ABSOLUTE_CONTROL)? Could you post the output of the `./cameractrls.py -l`?

```
❯ python cameractrls.py  -l
Basic / Crop
 zoom_absolute = 150    ( default: 110 min: 100 max: 400 )
 pan_absolute = 0       ( default: 0 min: -36000 max: 36000 step: 3600 )
 tilt_absolute = 0      ( default: 0 min: -36000 max: 36000 step: 3600 )
Basic / Focus
 focus_automatic_continuous = 0 ( default: 1 min: 0 max: 1 )
 focus_absolute = 115   ( default: 1 min: 1 max: 450 )
Exposure / Exposure
 auto_exposure = manual_mode    ( default: aperture_priority_mode values: manual_mode, aperture_priority_mode )
 exposure_time_absolute = 156   ( default: 156 min: 3 max: 2047 )
 exposure_dynamic_framerate = 1 ( default: 0 min: 0 max: 1 )
 gain = 29      ( default: 0 min: 0 max: 255 )
Exposure / Dynamic Range
 backlight_compensation = 1     ( default: 0 min: 0 max: 1 )
Color / Balance
 white_balance_automatic = 0    ( default: 1 min: 0 max: 1 )
 white_balance_temperature = 3490       ( default: 5000 min: 2800 max: 7500 step: 10 )
Color / Color
 brightness = 128       ( default: 128 min: 0 max: 255 )
 contrast = 128 ( default: 128 min: 0 max: 255 )
 saturation = 128       ( default: 128 min: 0 max: 255 )
 sharpness = 128        ( default: 128 min: 0 max: 255 )
Advanced / Power Line
 power_line_frequency = 50_hz   ( default: 60_hz values: disabled, 50_hz, 60_hz )
Advanced / Processing
 kiyo_pro_ultra_nr_2d = None    ( values: off, on )
 kiyo_pro_ultra_nr_3d = None    ( values: off, on )
Advanced / Other
 kiyo_pro_ultra_distortion_correction = None    ( values: off, on )
Capture / Capture
 pixelformat = NV12     ( values: YUYV, MJPG, H264, NV12 )
 resolution = 640x480   ( values: 640x480, 640x360, 1280x720, 1920x1080, 2560x1440, 3840x2160 )
 fps = 30       ( values: 60, 30, 24, 20, 15, 10, 5 )
Capture / Info
 card = Razer Kiyo Pro Ultra
 driver = uvcvideo
 path = /dev/video0
 real_path = /dev/video0
Settings / Save
 systemd_save           ( buttons: save )
```

pohlt, 2023-06-09:
* All of the exposure-related stuff seems to work, but depending on manual or auto mode, some parameters should be disabled (I split the exposure stuff in two section: auto, manual).
* Focus-related parameters seems to work.
* Advanced parameters are fine, too, but "Lens Distortion Compensation" need a stream restart to show an effect.

There's still some HDR stuff missing, but we've covered about 90% now I guess. 🥳
This is great. Thanks a lot!

One thing I don't understand: The automatic white balance is _way_ too warm. I always have to set it manually to about 4500 to not look like I'm sitting in front of a crackling bonfire. AWB works fine in Win 11. 🤔

soyersoyer, 2023-06-09:
It is interesting, that it has an H264 encoder,  but  the specification page doesn't mention it.
The specs says it has 82°- 72° field of view. Does it have a control for it in Synapse? Or only 640x480 has 72° FoV?
Could you post the output of the `v4l2-ctl --list-formats-ext`?

What happens if you change the exposure_time_absolute? Does it have the same effect as shutter speed? (the former is in 100 µs)

The nonfunctional AWB is interesting. Does it work outside synapse in Win?

pohlt, 2023-06-09:
The exposure time setting is weird:
* The image looks the same for any value between 3 and 2046.
* Only for 2047 the image is much brighter (actually too bright).

The shutter time setting behaves much more predictable and as expected.

The two settings seem to influence each other:
* If I change the exposure time to any value between 3-2046, the shutter time seems to be set implicitly to a value of about 30,000.
* If I set the exposure time to 2047 (image too bright), it seems to be reset to a value between 3-2046 when I set any shutter time value.

soyersoyer, 2023-06-09:
I've added a better slider for the shutter speed.

pohlt, 2023-06-09:
Save seems to be c003a80000000000.
Right before the save, there's a c0090a0000000000.

Even if I don't change anything, there are often c009060000000000 and c009080000000000 in the mix.

soyersoyer, 2023-06-09:
Save added.
Do they return anything? There is a query if the control parameter is 0x02, (not 0x01).

Bleuzen, 2023-09-07:
Hi, what is the current state of this?
Any help needed?

pohlt, 2023-09-07:
I'm currently too busy to contribute so if you own a Kiyo feel free to fill in the gaps. There is some basic support, though.

soyersoyer, 2023-09-07:
Did the save work?

davidhealey, 2024-12-24:
Has any of this been merged? I'm looking to buy one of these cameras.

pohlt, 2025-01-06:
> Has any of this been merged? I'm looking to buy one of these cameras.

Most (all?) of the camera's functionality is supported. I'm using it several times a week. Great camera.

agluck91, 2025-01-30:
Is there anything pending on this, I am planning to buy one of these. Ready to up my camera game but not ready to drop 600 dollars on an entry dslr yet. I can help if there is anything missing and you want to work on it.

f0ns, 2026-10-07:
Some findings from a Kiyo Pro Ultra (1532:0E08) on macOS, using the commands from this thread over IOKit, verified by grabbing frames before/after each change:

- Works: mirror (`c00e03`), exposure compensation (`c00e05`, −2 EV vs +2 EV clearly visible), metering (`c00e04`), ISO (`c00901`, ISO 100 vs 1600), shutter (`c00905`, 1/500 vs 1/30), 3D NR (`c00e01`, measurably less noise), AF mode/lighting (`c00a01`).
- The `c0 ..` commands echo the applied value on selector 2, e.g. mirror on → `00 00 00 01 00 00 00 00`, shutter 1/30 → `00 00 00 00 00 82 35 00`. Handy to confirm a write. `ff ..` commands reply all zeros.
- Lens distortion (`ff 01 0x 03`): no visible change with a fresh stream; seems to need a camera restart as noted above.
- Kiyo Pro HDR (`ff 02 0x`) is accepted (reply `80 ..`) but has no visible effect on the Ultra, so Ultra HDR is still unknown.
- Synapse's field of view appears to follow the digital zoom: it shows "FOV 44°" at 2× zoom, consistent with a ~78° full width.

Thanks for documenting these, it made a macOS app for this camera possible.

I'll share it on my Github after some final testing.

f0ns, 2026-10-07:
Here it is: https://github.com/f0ns/kiyo-control