soyersoyer/cameractrls #20: [Feature Request] Virtual camera

soyersoyer, 2023-04-27:
Here I collect feature requests that cannot be done with standard webcams.
But it can made by processing the camera stream and feeding it to a virtual camera.

 - Add [Green/Magenta] Tint
 - Add ability to load a LUT
 - Flip horizontally and vertically (even if the camera doesn't support it)
 - Rotate
 - Add background blur

This is out of scope yet.

seboss666, 2023-10-17:
Being able to have a virtual background image/blur is something that can be missing from multiple communication applications, specially under Linux when you have absolutely no camera control software from the vendors. But not sure how hard it can be on hardware for the image processing :/

v1r4j, 2024-06-04:
Is there a suggested workaround to flip the webcam image 180 degrees (flip vertically) for webcams where this functionality is not supported in v4l?

soyersoyer, 2024-06-04:
What kind of webcam do you have? The preview window can mirror/flip the image with m, and rotate with r, but it doesn't work outside the preview.

v1r4j, 2024-06-05:
It is a Lenovo LC50. It's housed upside down due to my set up and I am looking for a way to flip it that persists on video chat apps (google meets etc.)

soyersoyer, 2024-06-05:
Try OBS Studio's [virtual camera ](https://obsproject.com/kb/virtual-camera-guide) or [Webcamoid](https://webcamoid.github.io/)