soyersoyer/cameractrls #22: [Feature Request] Pixel rotation

vanarebane, 2023-05-10:
Hi,

I have a tiny USB type-c camera, with no wire, sits on the usb slot and the camera is in 90 degree angle.

Would be possible to add option/feature to rotate the preview by 0°, 90°, 180° and 270° ?

Thanks! Great app!

soyersoyer, 2023-05-15:
Hi,
Good to hear, that this works with a thermal camera as well.
I've added option to rotate or mirror the preview image. In the preview app you can rotate by pressing the r key, or mirror with the m key. Or you can also set these with the cli, in case if you want to make a runner for them.
It will be available in the next version. Or immediately if you clone the repo.
What pixel formats does your camera support? Which one works?

soyersoyer, 2023-06-14:
In v0.5.2 I also added colormaps, if your camera uses GRAY format.
Feel free to reopen this, if you have more questions.

closed 2023-06-14 by soyersoyer

vanarebane, 2023-06-24:
Hi, sorry for long wait for reply. It was a long vacation. Thanks for implementing the changes. I updated my app to version 5.6. 
Your app gives only YUYV pixel format option for the thermal camera. There is not GRAY format.
Rotation works, but it seems to be not resizing the preview window. So if the video is rotated to portrait and the video is bigger than the preview screen (top and bottom is cut off and left and right sides are black) and the video appears to be square.

I'm not sure if I know how to clone the repo, to test out new features faster. On Steam Deck it's a bit limited.

I made some hasty pictures of the Steam Deck using a thermal camera. I can try later to get better pictures if you like to use some to show off your app. https://1drv.ms/f/s!ApRLRqw9UoyuutUkTrQ9hbqpJRFiJA?e=86ae9f

soyersoyer, 2023-06-24:
Looks hot! :) Infiray P2 pro is your camera?
Is there a virtual keyboard or how to change the rotation in steam deck?
The rotation bug was fixed in v0.5.7. Going to flathub soon.
I'll figure out how to add some color to the yuyv thermal stream also.

vanarebane, 2023-06-27:
Yes, Infiray P2 pro camera!
I use the Steam Decks own virtual keyboard yes.

soyersoyer, 2023-06-27:
Use 'c' button to change the colormap to better show the hot fields. In the future I will add a minimal GUI for the preview.