soyersoyer/cameractrls #26: Make it work with Apple's AVFoundation

retpolanne, 2023-06-09:
Hey folks! 

It would be nice if this app could also be, whether through different backends implemented or a whole fork, supported by macOS. I use mostly mac for work with a Kiyo Pro camera and I wanted HDR so badly. 

I wonder if AVFoundation has the capabilities that v4l2 has so we can enable HDR and FOV.

soyersoyer, 2023-06-09:
It is possible to write controls with AVFoundation backend and use the GTK frontend as is, and enable controls whether AVFoundation or V4L2 is available, maybe. But I don't use a Mac.
If you want to use HDR or change FoV, you can install a Linux on a virtual machine, install this app, redirect usb, set the controls, and save it to the Kiyo Pro. Then it will be enabled on Mac as well.

retpolanne, 2023-06-09:
Thanks for the idea! I can try with multipass, I believe. 

@soyersoyer I see you made a comment on this issue here as well https://github.com/Itaybre/CameraController/issues/33 . I'll see if the CameraController owner can share the branch where they implemented HDR so I can try to implement what we have here for HDR and FoV settings.

soyersoyer, 2023-06-14:
Itaybre will implement the Kiyo Pro controls in the [CameraController](https://github.com/Itaybre/CameraController). We can close this here :)

closed 2023-06-14 by soyersoyer