soyersoyer/cameractrls #31: Manual exposure stopped working

trapexit, 2023-10-12:
v0.5.10 via flatpak

I apologize that I don't have more details on what changed on my side but the "Exposure Time, Absolute" slider in Cameractrls has become grayed out / fixed for my Anker PowerConf C200. Might have come after a kernel update.

[image]

[image]

soyersoyer, 2023-10-12:
Could you post the output of the `v4l2-ctl -L`?

trapexit, 2023-10-12:
```
User Controls

                     brightness 0x00980900 (int)    : min=0 max=100 step=1 default=50 value=50
                       contrast 0x00980901 (int)    : min=0 max=100 step=1 default=50 value=50
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=50 value=50
                            hue 0x00980903 (int)    : min=0 max=3 step=1 default=1 value=0
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=0 max=800 step=1 default=400 value=400
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=0 value=2 (60 Hz)
				0: Disabled
				1: 50 Hz
				2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2300 max=6500 step=1 default=4500 value=5333 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=100 step=1 default=50 value=50

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=0 value=0
				1: Manual Mode
         exposure_time_absolute 0x009a0902 (int)    : min=1 max=1000 step=1 default=156 value=156 flags=inactive
                   pan_absolute 0x009a0908 (int)    : min=-36000 max=36000 step=3600 default=0 value=-10800
                  tilt_absolute 0x009a0909 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                 focus_absolute 0x009a090a (int)    : min=300 max=650 step=1 default=300 value=300 flags=inactive
     focus_automatic_continuous 0x009a090c (bool)   : default=1 value=1
                  zoom_absolute 0x009a090d (int)    : min=100 max=400 step=1 default=100 value=100
```

soyersoyer, 2023-10-13:
Does your camera have the latest firmware?

This looks suspicious:
```
auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=0 value=0
				1: Manual Mode
```

There is no menu item for value 0, and this is not handled well yet. I'll create a fix. Thanks!

Remember an earlier kernel had 'Aperture Priority Mode' for auto exposure control? (Not only 'Manual Mode')

trapexit, 2023-10-13:
> Does your camera have the latest firmware?

I'll check. I haven't changed anything with the camera since I got it. I'll check it.

> Remember an earlier kernel had 'Aperture Priority Mode' for auto exposure control? (Not only 'Manual Mode')

I can try the camera elsewhere or roll back to check but I don't recall that.

soyersoyer, 2023-10-13:
0.5.11 has been released on flathub.

trapexit, 2023-10-13:
Looks to be working. Thank you for the quick turnaround.

closed 2023-10-13 by trapexit