soyersoyer/cameractrls #34: Feature Request: H264 support

jcfrosty, 2023-11-16:
Requesting additional H264 support for high end cameras.
[image]

soyersoyer, 2023-11-20:
Nice specs :) 4k60p H264 decoding could be tough.
Could you please check if 4k60p H264 works with guvcview? It uses FFmpeg/AVCodec, if it's fast enough I can add a similar solution.

jonathanhirschman, 2023-12-29:
H264 at lower resolutions and FPS would be super useful, even if  the software decoders can't keep up :smile:

nekohayo, 2025-01-23:
Unless you are talking about something else here, I am under the impression that if you were to use PipeWire / XDG Camera for access (#71) and [GStreamer's GTK4 paintable sink](https://gstreamer.freedesktop.org/documentation/gtk4/index.html?gi-language=python), you would benefit from not only video acceleration, but also zero-copy. @rmader can correct me if I'm misremembering or missing something :)

rmader, 2025-01-23:
Uh, I don't think Pipewire would be a easy fit here as the app seems to be about low level control. Pipewire is an abstraction on top of v4l2 and libcamera - and might not cover all functionality yet. Even though adding missing stuff is usually quite easy and it would make it easy for the app to also support libcamera.

In any case, it would be a big architectural change.

nekohayo, 2025-01-27:
Ah, in #71 I meant using PipeWire for access _for video/image preview_ and GStreamer's sink for actually displaying the preview, while still using V4L for controlling the parameters otherwise. I have not seen the parameters setting as conflicting / blocking other apps from accessing the webcam's video feed.