soyersoyer/cameractrls #35: Cannot save to Systemd

richieri, 2023-11-24:
Hello!

I'm getting the following error from cameractrls.cameractrlsgtk when trying to click on Save (save settings to systemd):

`Traceback (most recent call last):
  File "/snap/cameractrls/31/usr/bin/cameractrlsgtk.py", line 340, in <lambda>
    b.connect('clicked', lambda e, c=c, m=m: self.update_ctrl(c, m.text_id))
  File "/snap/cameractrls/31/usr/bin/cameractrlsgtk.py", line 429, in update_ctrl
    self.camera.setup_ctrls({ctrl.text_id: value}, errs)
  File "/snap/cameractrls/31/usr/bin/cameractrls.py", line 2244, in setup_ctrls
    c.setup_ctrls(params, errs)
  File "/snap/cameractrls/31/usr/bin/cameractrls.py", line 2131, in setup_ctrls
    self.create_systemd_service_and_path()
  File "/snap/cameractrls/31/usr/bin/cameractrls.py", line 2155, in create_systemd_service_and_path
    subprocess.run(["systemctl", "--user", "enable", "--now", service_file])
  File "/usr/lib/python3.10/subprocess.py", line 503, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1863, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
PermissionError: [Errno 13] Permission denied: 'systemctl'
`

Ubuntu 22.04
Version: 0.5.11-0
Installed from Ubuntu Software with snap

mikegleasonjr, 2024-02-20:
Same, some more logs regarding `apparmor`:

```
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]: Traceback (most recent call last):
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]:   File "/snap/cameractrls/31/usr/bin/cameractrlsgtk.py", line 340, in <lambda>
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]:     b.connect('clicked', lambda e, c=c, m=m: self.update_ctrl(c, m.text_id))
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]:   File "/snap/cameractrls/31/usr/bin/cameractrlsgtk.py", line 429, in update_ctrl
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]:     self.camera.setup_ctrls({ctrl.text_id: value}, errs)
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]:   File "/snap/cameractrls/31/usr/bin/cameractrls.py", line 2244, in setup_ctrls
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]:     c.setup_ctrls(params, errs)
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]:   File "/snap/cameractrls/31/usr/bin/cameractrls.py", line 2131, in setup_ctrls
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]:     self.create_systemd_service_and_path()
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]:   File "/snap/cameractrls/31/usr/bin/cameractrls.py", line 2155, in create_systemd_service_and_path
Feb 20 10:27:52 computer kernel: [  851.676752] audit: type=1400 audit(1708421272.881:119): apparmor="DENIED" operation="exec" class="file" profile="snap.cameractrls.cameractrlsgtk" name="/usr/bin/systemctl" pid=17448 comm="python3" requested_mask="x" denied_mask="x" fsuid=1000 ouid=0
Feb 20 10:27:52 computer kernel: [  851.676911] audit: type=1400 audit(1708421272.881:120): apparmor="DENIED" operation="exec" class="file" profile="snap.cameractrls.cameractrlsgtk" name="/usr/bin/systemctl" pid=17448 comm="python3" requested_mask="x" denied_mask="x" fsuid=1000 ouid=0
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]:     subprocess.run(["systemctl", "--user", "enable", "--now", service_file])
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]:   File "/usr/lib/python3.10/subprocess.py", line 503, in run
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]:     with Popen(*popenargs, **kwargs) as process:
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]:   File "/usr/lib/python3.10/subprocess.py", line 971, in __init__
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]:     self._execute_child(args, executable, preexec_fn, close_fds,
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]:   File "/usr/lib/python3.10/subprocess.py", line 1863, in _execute_child
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]:     raise child_exception_type(errno_num, err_msg, err_filename)
Feb 20 10:27:52 computer cameractrls_cameractrlsgtk.desktop[17294]: PermissionError: [Errno 13] Permission denied: 'systemctl'
```

Slater91, 2024-06-26:
I spent a bit trying to figure this out. In `/var/lib/snapd/apparmor/profiles/snap.cameractrls.cameractrls` as well as in `/var/lib/snapd/apparmor/profiles/snap.cameractrls.cameractrlsgtk` add at the end, before the last `}`:
` /usr/bin/systemctl rix,`

After saving the file, run `sudo apparmor_parser -r /var/lib/snapd/apparmor/profiles/snap.cameractrls.cameractrls*`. It should now work. In my case, the application was not even starting anymore because it was blocked by AppArmor, but now it works. The logs don't have any more complaints from apparmor for trying to use systemctl. Unfortunately it doesn't work anyway and I can't get anything out of running the application from the terminal (it simply launches and immediately returns without writing anything to the console).

shanness, 2024-07-31:
Thanks so much @Slater91 , been banging my head against the wall for ages with this one!

RokeJulianLockhart, 2024-11-07:
> trying to click on Save

Where is that option, @richieri? I'd like to evaluate whether this works on Fedora 41, but haven't been able to ascertain what preferences can be saved to SystemD.