soyersoyer/cameractrls #44: Feature Request: Dark mode (Gnome 45)

JajinderK, 2024-03-21:
Hi there. Love the ease of use. Can you please add a dark mode feature? In the "gnome-software" I see (very blurry) screenshots where the application is dark, but once after installment it doesn't go dark, nor can I find an option. I'm on Fedora, Gnome 45. Thanks in advance.

soyersoyer, 2024-03-21:
Hi JRKL75, Please install the gnome-tweaks, and set Appearance/Styles/Legacy Applications to Adwaita-dark. If you are using the Flatpak version, you will probably need to install org.gtk.Gtk3theme.Adwaita-dark as well.

JRKL75, 2024-03-21:
Hi Thanks for reply. I'm on Adwaida dark mode already.
Version GTK 3.24.41
Tweaks -> Legacy Applications -> Adwaita (default)

soyersoyer, 2024-03-21:
You will probably also need the gnome-themes-extra package. It includes gtk3 adwaita-dark theme.

JRKL75, 2024-03-21:
Thanks for the tip. It added the Adwaita-dark, but unfortunately not changing. Settings it to "HighContrastInverse" makes it dark though, but it's not looking that great XDXD. No idea what's causing it to not function. Ah well, we tried. Thanks anyway.

soyersoyer, 2024-03-21:
HighContrastInverse is very ugly.
I think it should work if you install the Adwaita dark theme into the Flatpak environment as well. You can do that from Software app, or `flatpak install org.gtk.Gtk3theme.Adwaita-dark`, or just reinstall cameractrls while Adwaita-dark is selected in Tweaks.