soyersoyer/cameractrls #46: Can the settings be saved?

skorokithakis, 2024-04-02:
The camera keeps forgetting the settings very often, most importantly the resolution. This utility is a great help, as I can set them using an alias, but is there a way for the settings to be persisted on the camera?

soyersoyer, 2024-04-02:
Yes, if you enable cameractrlsd in the Advanced page, it will restore the 'preset 1' at every device connection or every login.
Does it work for you?

skorokithakis, 2024-04-02:
Ahh ok, I need to run the daemon? Makes sense, thank you. I was wondering if there's a way to save the settings on the hardware, but this works as well. I'll see about properly installing the app (I just copied the cli script and use it directly, currently).

soyersoyer, 2024-04-02:
Some Logitech cameras save the settings, Kiyo Pro has a control for it, but not all can save. It will work if you use from the cli or package (eg on arch) or from flatpak as well.

skorokithakis, 2024-04-02:
I've installed the daemon, thank you. I have a Brio 500, I'm assuming it can't save the settings. Thanks again!

ItachiSan, 2024-04-03:
In Arch they are not packaging the systemd daemon script, most likely due to a simple mistake.
I opened an issue so that they can fix it:
https://gitlab.archlinux.org/archlinux/packaging/packages/cameractrls/-/issues/3

ItachiSan, 2024-04-04:
Hi,
The daemon has been added to the Arch package and is up and running.
However, clicking on a preset does not lead to saving it. I get the error in the terminal that the .ini file is missing.
@soyersoyer

soyersoyer, 2024-04-04:
Click to load, long press to save. If you move the cursor over the name of the controls, the description will be displayed. Hm, maybe it should be more visible.

soyersoyer, 2024-04-04:
Or should this "auto-radio preset style" be changed to something that exists in the world today.

ToRRent1812, 2024-04-04:
Pardon my stupidity but in flatpak version, whenever I click enable or disable for 'start with desktop portal', nothing is checked but the app itself launches automatically. I also thing that the part where You need to hold click to save is confusing, considering, there is no tooltip on hover to know that.

soyersoyer, 2024-04-04:
Ups, maybe the desktop portal is not implemented the same way anywhere.
Which desktop environment are you using? (KDE/Gnome/Xfce/or other)?

ToRRent1812, 2024-04-04:
I use Plasma 6.0.3, distro is Opensuse tumbleweed

soyersoyer, 2024-04-04:
I'll check it out in the next few days.

ItachiSan, 2024-04-05:
I got the save preset to work, by long pressing a fairly long time.
However, I got no feedback that it was correct.
Maybe that could be improved?

jimmysjolund, 2024-04-05:
I was also confused by the long press when I couldn't the presets to work. I run Fedora 39 with Gnome and no tooltip on hover.

rieje, 2024-04-05:
Is it possible to support saving settings without requiring a daemon to a simple config file? I rarely use the webcam. That would seem to be the simplest and most intuitive solution.

soyersoyer, 2024-04-05:
Yes saving and loading work without the daemon. The daemon is only needed to restore controls.

You can also restore the controls with udev, but this requires root privileges.
https://wiki.archlinux.org/title/webcam_setup#Persisting_configuration_changes

rieje, 2024-04-08:
> Yes, saving and loading work without the daemon. The daemon is only needed to automatically restore controls.
> 
> You can also restore the controls with udev, but this requires root privileges. https://wiki.archlinux.org/title/webcam_setup#Persisting_configuration_changes

You mean it should remember the settings persistently? When I change some settings ad restart cameractrlsgtk4, it remembers them but they get reset to defaults on reboot (I would prefer if they get remembered). I guess I can do `cameractrls -c logitech_led1_mode=off,resolution=1280x960 ; cameractrlsgtk4` in a wrapper script if I don't want to bother with udev rule or a daemon.

shanness, 2024-08-12:
> I was also confused by the long press when I couldn't get the presets to work. I run Fedora 39 with Gnome and no tooltip on hover.

Whoa!  Thanks so much for this!  It works and there is a tooltip on Ubuntu 22.04, flatpak, I'd just never noticed it.
 (the snap seem broken ATM, needs root for a systemctl exec command)
 
 I think the long click should be in the label?
 
[image]

hensansi, 2024-09-12:
For those here who use Sway, the "Start with systemd" option won't work out of the box because it never reaches "graphical-session.target."[1] If you run `systemctl status cameractrlsd.service --user`, it will show enabled but inactive.

To address that, you can follow this guide: [4.10 Manage Sway-specific daemons with systems](https://wiki.archlinux.org/title/Sway)

```
~/.config/systemd/user/sway-session.target

[Unit]
Description=Sway compositor session
Documentation=man:systemd.special
BindsTo=graphical-session.target
Wants=graphical-session-pre.target
After=graphical-session-pre.target
```
```
~/.config/sway/config

...
...
...
exec_always systemctl --user start sway-session.target
```

[1] To give some extra context, you can check [this issue](https://github.com/swaywm/sway/issues/7862).