soyersoyer/cameractrls #48: Logitech BRIO 4K - resolution, fps ?

mstup, 2024-04-10:
Hi,

Looks like there are no ways to set highest supported resolution and fps for Logitech BRIO 4K camera (4K, 30fps, 2k 60fps) on Fedora 38.
Or it's driver problem ?

```
./cameractrls.py -l
... 
Capture / Capture
 pixelformat = MJPG	( values: YUYV, MJPG )
 resolution = 1920x1080	( values: 1920x1080, 1600x896, 1280x720, 1024x576, 960x540, 848x480, 800x600, 800x448, 640x480, 640x360, 480x270, 424x240, 352x288, 320x240, 320x180, 176x144, 160x120 )
 fps = 30	( values: 30, 24, 20, 15, 10, 7.5, 5 )
```

mstup, 2024-04-10:
No, USB 3

soyersoyer, 2024-04-10:
What does the lsusb / lsusb -t says? Your cable is probably only for USB2.

mstup, 2024-04-10:
```
lsusb -t
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 003: Dev 002, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 003: Dev 002, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/:  Bus 003.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/3p, 480M
    |__ Port 003: Dev 004, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 003: Dev 004, If 1, Class=Wireless, Driver=btusb, 12M
/:  Bus 004.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/:  Bus 005.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/1p, 480M
/:  Bus 006.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/0p, 5000M
/:  Bus 007.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/1p, 480M
/:  Bus 008.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/1p, 10000M
/:  Bus 009.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 001: Dev 002, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 001: Dev 002, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 001: Dev 002, If 2, Class=Video, Driver=uvcvideo, 480M
    |__ Port 001: Dev 002, If 3, Class=Audio, Driver=snd-usb-audio, 480M
    |__ Port 001: Dev 002, If 4, Class=Audio, Driver=snd-usb-audio, 480M
/:  Bus 010.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/1p, 10000M
```

mstup, 2024-04-10:
@soyersoyer yes, probably you are right. Thank you for clarification.

soyersoyer, 2024-04-10:
This is a USB2 bus.
`Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

mstup, 2024-04-10:
This is my USB cable limitation. Thanks you.

soyersoyer, 2024-04-10:
You are welcome!

closed 2024-04-10 by soyersoyer