soyersoyer/cameractrls #49: Cameractrls can put you camera in a barely recoverable state

gbin, 2024-05-03:
I had issues with 2 cameras:
- one with the Razer Kiyo ultra pro, there are combinations of settings that can after you set it up in the all and you use it with a standard UVC client (like chrome) put the camera is a crashed/frozen state (light stay on, you cannot connect to it anymore etc.) I could recover by installing windows, going into windows, installing the razer driver, override parameters and go back to linux.

- by mistake I actually used cameratrls on my laptop camera (on the Framework Laptop) and now it is permanently dark whatever setting I twiddle, I haven't found a solution yet to recover from it.

I believe you need to be sure in this software you can recover good, known comprehensive settings.

soyersoyer, 2024-05-05:
I'm sorry you experienced this. I haven't come across anything similar yet, although I can't test it with every camera. The application only sets V4L2 controls for most cameras, which other applications can also set. The problem is probably in the firmware of your cameras.

gbin, 2024-05-05:
Ok I understood what was going on fully. This is not Cameractrls entire fault. I was using mainly google meet to test the camera and it looks like meet is buggy, if you reset meet (remove the chrome app storage) it will reread the parameter from the camera and land back on its feet.

closed 2024-05-05 by gbin