soyersoyer/cameractrls #55: Insta 360 Link Support

jorgecastro05, 2024-06-23:
The Insta 360 Link webcam, works as expected with gimbal, brightness, contrast, saturation and sharpness controls, but is missing the controls related to exposure, the official application for windows systems support manual and automatic exposure, so I would like to add these controls with cameractrls.

Is there any developer guide, or  starting point to find the exposure controls, not available using the v4l2-ctl, or some way to search these parameters, and adding manually to cameractrls?

lsusb output

    Bus 001 Device 010: ID 2e1a:4c01 Insta360 Insta360 Link

v4l2-ctl output

```
v4l2-ctl --device=/dev/v4l/by-id/usb-Insta360_Insta360_Link-video-index0 -L

User Controls

                     brightness 0x00980900 (int)    : min=0 max=100 step=1 default=50 value=50
                       contrast 0x00980901 (int)    : min=0 max=100 step=1 default=50 value=50
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=50 value=50
                            hue 0x00980903 (int)    : min=-15 max=15 step=1 default=0 value=0
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=3 value=2 (60 Hz)
				0: Disabled
				1: 50 Hz
				2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2000 max=10000 step=1 default=6400 value=5500 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=100 step=1 default=50 value=50

Camera Controls

                   pan_absolute 0x009a0908 (int)    : min=-522000 max=522000 step=3600 default=0 value=-4320
                  tilt_absolute 0x009a0909 (int)    : min=-324000 max=360000 step=3600 default=0 value=-277920
                 focus_absolute 0x009a090a (int)    : min=0 max=100 step=1 default=50 value=0 flags=inactive
     focus_automatic_continuous 0x009a090c (bool)   : default=0 value=1
                  zoom_absolute 0x009a090d (int)    : min=100 max=400 step=1 default=100 value=100
```

soyersoyer, 2024-06-24:
Try to capture the USB communication with Wireshark/USBPcap. If you are lucky, it uses an UVC Extension unit to set exposure controls. Then you can add a new extension like the [KiyoProCtrls](https://github.com/soyersoyer/cameractrls/blob/main/cameractrls.py#L1174) class. Feel free to ask if you get stuck, I can help.

dan9thsense, 2024-10-03:
Did you have success with this? I am trying something similar for the Insta 360 X3 camera. Specifically, that camera comes up in webcam mode with a split screen. A button press on the camera toggles that to forward camera, rear camera, split screen. Is there any way to know if that toggle can be controlled via webcam controls?

jorgecastro05, 2025-04-24:
Sorry for the late response, now that I have some time for test, I configured Wireshark/USBPcap, open instalink 360 control software and captured packages, and seems like I found the controls for exposure, (thanks to deepseek for help with analyzing packets and general help), in summary I followed the next steps:

1. Filter for UVC control transfers (most exposure commands use Control Transfers):

`usb.transfer_type == 2 && usb.bmRequestType == 0x21`

2. Look for "SET_CUR" requests, send to deepseek to analyze with this result:

captured SET_CUR (Set Current) requests sent to Unit 9, Control Selector 0x19, which appear to control exposure settings

| **Packet No.** | **Exposure Value (Decimal)** | **Hex Payload (Little-Endian)** |                              **Meaning**                             |
|:--------------:|:----------------------------:|:-------------------------------:|:--------------------------------------------------------------------:|
| 261            | 400                          | 90 01 (0x0190)                  | Sets exposure to 400 (likely in microseconds or an arbitrary scale). |
| 417            | 500                          | F4 01 (0x01F4)                  | Sets exposure to 500.                                                |
| 427            | 640                          | 80 02 (0x0280)                  | Sets exposure to 640.                                                |

That was correct since I changed the exposure of the cam with those values. (400, 500 and 640)

3. Then the following response was for control exposure:

### **2. UVC Extension Unit Structure**

-   **`bmRequestType: 0x21`**
    
    -   Direction: Host → Device (`0x00`)
        
    -   Type: Class (`0x01`)
        
    -   Recipient: Interface (`0x01`)
        
-   **`bRequest: 0x01`** (`SET_CUR`)
    
    -   Standard UVC request to **set a value**.
        
-   **`wValue: 0x1900`**
    
    -   **Control Selector = `0x19`** (likely a custom exposure control for this camera).
        
    -   High byte (`0x19`) = selector, low byte (`0x00`) = unused.
        
-   **`wIndex: 0x0900`**
    
    -   **Unit ID = `0x09`** (Extension Unit 9).
        
-   **Payload**: `90 01` (little-endian `0x0190` = `400`).

now I will try to create a simple python script for test the control of the webcam, and hopefully start to create the extension like you suggested.

dan9thsense, 2025-04-24:
This is great progress, thanks. Looking forward to the python script

jorgecastro05, 2025-04-24:
> And ideas for toggling the split screen?

if you can toggle that function from a computer, or laptop with the camera connected via USB, probably the same steps looking the controls by Wireshark/USBPcap could work, as I did with this one. if it's a function only available from the a button in camera itself I'm not sure if could be possible toggle that function.