soyersoyer/cameractrls #56: Resolution Drops!

sallyhaj, 2024-07-02:
Hello there,
Thank you for the amazing application.
I am using UGREEN camera 2K on Debian 12. When I open Cameractrls and change the resolution to like 1920x1080 I can preview that without an issue, even with 2592x1520.
The problem is that, when I want to make it in real use, for example if I open jitis or google meet, once I open the camera there, the resolution drops to 1024x768, even I open Cameractrls at the same time and change the resolution, but it drops immediately.

What do you think might cause this?
Thank you.

soyersoyer, 2024-07-09:
Probably Jitis or Google Meet changed the resolution. Has the Pixel format been changed as well?

sallyhaj, 2024-07-09:
The Pixel is set to MJPG, if I change it to YUYV and open Jitis/Google Meet, it changes automatically to MJPG.

aleixq, 2025-01-09:
It happens to me too with jitsi , as @soyersoyer  said, it seems that jitsi is lowering the resolution, to solve in jitsi side : add `#config.resolution=1080` to the jitsi url:  https://github.com/jitsi/jitsi-meet/issues/9648 .