soyersoyer/cameractrls #58: Better control over sliders' precision granularity, and SpinButton widgets to set exact numeric values (not just sliders)

nekohayo, 2024-08-13:
I love using this app to manually adjust and lock the exposure, white balance and focus on my webcams (a Logitech C920 and a C270).

However, in some lighting conditions, it is extremely difficult to do so, because the app only presents me with sliders, and the values I need are always in the lower 1% of the slider's range, so it's almost impossible to get a precision adjustment done with the mouse (and the keyboard moves values way too slowly).

I'd need a way to be able to set the minimum/maximum values ranges for those sliders or, probably more simply, to have an accompanying GtkSpinButton widget next to them where I can just enter and adjust the exact value I want, instead of fighting with the slider.

Or, for example, there are cases where I know the exact value I need, such as white balance temperature: I might know that by spec, my lighting is 5500K ±100, or that I'm using an incandescent bulb, or D65 lighting, or… so I'd want to be able to input exactly `5500`, for example.

Here's an example of how the current GUI is nearly impossible to use for correct exposition (in this example I'm trying to expose to have no blown highlights on the white sheet of paper, with a Logitech C270):

https://github.com/user-attachments/assets/1fd0ad72-47cd-4e14-b529-a8a1e505c7b2

closed 2024-08-17 by soyersoyer

soyersoyer, 2024-08-17:
Hi Jeff, Thanks for this idea and the video. I added SpinButtons next to the Scales.

nekohayo, 2025-02-14:
Is this supposed to be included in the 0.6.7 release that was published in September? I'm not seeing it in the UI in that version.

soyersoyer, 2025-02-27:
Yes, there is an arrow that allows you to switch to the spin button.

commit 0f19a637 (2024-08-17): cameractrlsgtk[4]: add SpinButtons next to Scales

Sometimes it is difficult to choose the right value on the scale. Add a SpinButton for better value adjustment.

Fixes: #58