soyersoyer/cameractrls #60: Flathub Verification

Roman19S, 2024-08-19:
Distros like Linux Mint and Elementary OS have started hiding or otherwise discouraging the use of unverified flatpaks. More will likely follow as verified apps approach 50% of Flathub. Could you verify the flatpak on Flathub so people on these distros can find it? Here is a link to Flathub's verification documentation for app authors https://docs.flathub.org/docs/for-app-authors/verification/

soyersoyer, 2024-08-19:
App verification only works with GitHub login, but it doesn't require the minimum set of information, so unfortunately I can't verify it yet.

I think this part is unacceptable:
'This application will be able to read your organization, team membership, and private project boards. '

Roman19S, 2024-08-19:
That is perfectly understandable. You can also contact someone at Flathub and ask them to manually verify it

gelbphoenix, 2024-09-20:
> App verification only works with GitHub login, but it doesn't require the minimum set of information, so unfortunately I can't verify it yet.

After the documentation it should also function with the `irl.hu` domain if that is your domain. That's because the app ID is `hu.irl.cameractrls`.

soyersoyer, 2024-09-20:
Yes, that's my domain.
Unfortunately, I can't start the verification process, because I'm logging in through GNOME Gitlab, not GitHub.

gelbphoenix, 2024-09-20:
Then you should maybe start a manual verification with the Flathub admins. Also this *could* be a bug on the side of Flathub.

soyersoyer, 2024-09-21:
If I were to start a manual verification, that would mean I agree with a system where developers give up their own privacy for a blue nike badge.

gelbphoenix, 2024-09-22:
> I think this part is unacceptable:
> 'This application will be able to read your organization, team membership, and private project boards.'

If this is your point your not accepting than this whole issue is null. That's because I IMO don't see any way where the Flathub team will be changing that for the GitHub login (if they are even possible to single out specific permissions of that group).

But that is better discussed on Flathub's infrastructure.

Roman19S, 2024-09-22:
I'll start a discussion on the Flathub forums about this flaw with verification. @soyersoyer thanks for the valuable feedback. Hopefully Flathub will be able to adjust the verification process to better fit your needs. I'll close this issue, at least until something changes with verification, to avoid cluttering up your issues.

closed 2024-09-22 by Roman19S