soyersoyer/cameractrls #62: FoV on Logitech MX Brio 4K

erdmann040, 2024-09-19:
Hi,

I'm using the version 0.6.6, installed via dnf on Fedora.

My Webcam is a "Logitech MX Brio 4K". 

Your software works great on my system with one exception: The GUI shows me three possible settings for FoV: 65°, 78° and 90° (where 65° is preselected). If I click on one of the other two values, an error message appears:

LogitechCtrls: failed to set logitech_brio_fov to 78, current value 65

Anything idea how to fix this? 

Thanks in advance.

[image]

soyersoyer, 2024-09-20:
Unfortunately I don't have an MX Brio camera.  FoV support for the MX Brio camera was added in #43. @wanderboessenkool cam you help?

wanderboessenkool, 2024-09-20:
It might be a permission issue on the USB device for the camera. I have this in `/etc/udev/rules.d/83-brio.rules`, I probably added that by hand some time in the past:

```
SUBSYSTEM=="usb", ATTR{idVendor}=="046d", ATTR{idProduct}=="0944",MODE="0666"
```

After adding that file run:

```bash
sudo udevadm control --reload-rules
```

and then re-plug your camera.

@soyersoyer: If this works we need some way to ship that rule and have it be added to the system if a MX Brio is detected. This might be especially hard for the Flatpak, so maybe a bit of documentation is enough.

erdmann040, 2024-09-20:
I have no idea what I did when I followed @wanderboessenkool 's instructions but it worked. Thank you both very much :)

closed 2024-09-20 by erdmann040