soyersoyer/cameractrls #64: MIDI controller mapping, how?

cjeholm, 2024-10-19:
I can not figure out how to remap my controllers knobs. My controller, a MPK Mini mk3, is recognized and works on the default settings of controlling zoom. I'd like to control the manual exposure, gain and white balance. I have a feeling that I'm missing something trivial.

Tested on gtk4/gtk3, qtile and Plasma on NixOs unstable.

soyersoyer, 2024-10-19:
Only pan/tilt/zoom/preset can be controlled with it for now. Controlling the other controls via MIDI is a good idea. I am wondering how to add it in a simple way. I might map them to some undefined MIDI CCs (85-90, 102-119).

cjeholm, 2024-10-19:
I see, thanks.

I have a suggestion, how about a button next to the MIDI device - "Learning mode" or similar. This enables a button on each control that after being pressed binds the first incoming MIDI event to that control. The software synth Helm has a similar way of binding controls and it works quite well.