soyersoyer/cameractrls #76: Show a more human-readable camera name (using brand and model) among choices

nekohayo, 2025-02-26:
Testing with version 0.6.7 from Flathub, in the meatballs menu, these are the camera names presented to me (my laptop's built-in camera vs the USB one):

[image]

However, this doesn't really tell the user what is what, especially "UVC Camera". On the same system, doing `lsusb | grep cam` gives me this much more meaningful name:

```
Bus 003 Device 007: ID 046d:0825 Logitech, Inc. Webcam C270
```

Is there a way cameractrls could fetch the "Logitech, Inc. Webcam C270" information and display that instead of "UVC Camera"? And maybe skip the ` (046d:0825)` string display unless there are multiple cameras named identically?

soyersoyer, 2025-02-27:
The problem is that the product string is missing from the C270 USB descriptor.

```
  iProduct                0 
```

So the UVC driver gives the name "UVC Camera (046d:0825)" here:
https://github.com/torvalds/linux/blob/dd83757f6e686a2188997cb58b5975f744bb7786/drivers/media/usb/uvc/uvc_driver.c#L2196

The lsusb uses the udev hwdb to get the "Logitech, Inc. Webcam C270" name.
https://github.com/gregkh/usbutils/blob/74ea925365306c77c215cf73f829723712ed685f/names.c#L212

/lib/udev/hwdb.d/20-usb-vendor-model.hwdb:
```
usb:v046D*
 ID_VENDOR_FROM_DATABASE=Logitech, Inc.

...

usb:v046Dp0825*
 ID_MODEL_FROM_DATABASE=Webcam C270
```

Unfortunately this hwdb is not available within Flatpak by default.

Even through the Camera portal, only the name "UVC Camera (046d:0825)" given by the driver is available.

It could be queried from udev, and flatpak's access could be extended. But maybe this is the only camera I have that does not have a product string in the USB descriptor.

nekohayo, 2025-02-27:
Oh! I didn't realize there was a problem with _this specific model_ of webcam... and indeed, with a C920 there is no problem.

"UVC" is a bit technical though. I wonder if doing a check on that fallback string (on the application side) to display "USB camera" instead in such situations could be more user-friendly, as "USB" is a term much more commonly used by people in everyday conversations? Otherwise, not much we can do I guess.

soyersoyer, 2025-02-27:
If only cameractrls calls it USB Camera, but all other programs still call it UVC Camera, then this may confuse users.