soyersoyer/cameractrls #79: Universal Joystick Control

1cupoverflow, 2025-03-29:
There is not currently a universal graphical interface for controlling motorized webcams or software to allow users to control a motorized webcam with a hardware device like a joystick or keyboard via arrow keys. Adding these features to CameraCTRLS would make the software much more useful for modern tracking webcams that are motorized and able to track the user. 

Bonus: Creating an extensible control interface that can be accessed from the BASH shell and used to program for routine movements or creating a framework for AI movement tracking plugins would be more sophisticated but potentially very useful. Programmable tracking would be invaluable to amateur astronomers and custom AI plugins would make consumer tracking webcams useful for indoor surveillance and industrial environments..

soyersoyer, 2025-03-29:
There is some support. Please read the [PTZ](https://github.com/soyersoyer/cameractrls?tab=readme-ov-file#ptz-controls) section of the README.

1cupoverflow, 2025-03-29:
> There is some support. Please read the [PTZ](https://github.com/soyersoyer/cameractrls?tab=readme-ov-file#ptz-controls) section of the README.

Keyboard control via arrow keys has been added. Game console joystick support added. MIDI device controls added. Python CLI is available. 

Support for joysticks, BASH scripting, machine learning tracking plugins and graphical PTZ interface missing. 10/10 this software has everything, perfect. Problem solved.

soyersoyer, 2025-03-30:
Joystick support would be nice. I'll get one from somewhere.
A better graphical PTZ interface would also be nice for touch screens. Currently there are sliders for pan/tilt.
You can call python from BASH.
Tracking plugins are out of scope yet.