soyersoyer/cameractrls #80: Runtime error with GTK client

msvec, 2025-03-31:
I'm getting this error with the current git version:

```
bash$ python3.11 cameractrlsgtk.py 
Traceback (most recent call last):
  File "/tmp/cameractrls/cameractrlsgtk.py", line 701, in do_activate
    self.window = CameraCtrlsWindow(application=self, title='Cameractrls')
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/cameractrls/cameractrlsgtk.py", line 50, in __init__
    self.refresh_devices()
  File "/tmp/cameractrls/cameractrlsgtk.py", line 204, in refresh_devices
    self.gui_open_device(idx)
  File "/tmp/cameractrls/cameractrlsgtk.py", line 223, in gui_open_device
    self.init_gui_device()
  File "/tmp/cameractrls/cameractrlsgtk.py", line 428, in init_gui_device
    c.gui_ctrls += b
TypeError: 'Button' object is not iterable

```

soyersoyer, 2025-03-31:
Thanks for reporting this. Nice catch. I pushed a fix.
Interestingly, this warning doesn't appear with newer python.
It also doesn't get added to the list.

```
a = []
b = Gtk.Button()
a += b
print(a)

[]
```

msvec, 2025-03-31:
BTW this could be related:

```
Traceback (most recent call last):
  File "/tmp/cameractrls/cameraview.py", line 8, in <module>
    from cameractrls import CameraCtrls, PTZController
  File "/tmp/cameractrls/cameractrls.py", line 2002
    match c.text_id:
          ^
SyntaxError: invalid syntax
```

msvec, 2025-03-31:
3.11

I got that when clicking on _Show preview window_

soyersoyer, 2025-03-31:
I replaced the matches.

The preview button starts cameraview.py as a new process. And this uses the #!/usr/bin/env python3 shebang.
Is it possible that this is launching a version prior to 3.10 on your machine?

msvec, 2025-03-31:
Oh, that explains it. My python3 points to 3.6.